Guilouz/Creality-Helper-Script-Wiki #345: [BUG]  K1C Camera Settings Control not possible to install

P-C-R, 2024-01-30:
### What happened?

Wanted to install. Get this error:

```
Type your choice and validate with Enter: 18

 ✗ Your camera version is not compatible!
```

### What did you expect to happen instead?

Install

### How to reproduce this issue?

Install on K1C with 1.3.3.5

Stoony1, 2024-01-30:
same on my k1

Guilouz, 2024-01-31:
It’s not an issue, that’s means you have the new model of webcam that is not compatible and can’t be adjusted.

Hammerfest, 2024-02-04:
Yes, and everyone with a K1 variant (K1C/Max) should message Creality support requesting they return controls of the camera module to allow for better picture quality with added lighting.