Guilouz/Creality-Helper-Script-Wiki #346: [BUG] <title>

Stoony1, 2024-01-30:
### What happened?

Timelapse not work with my k1

### What did you expect to happen instead?

That i can use Timelapse

### How to reproduce this issue?

Everything installed according to the instructions, but the camera cannot be found in Mainsail with the time-lapse settings in Fluid, but the camera itself works wonderfully and smoothly, only time-lapse recordings are not possible

### Additional information:

I have a question: I have given my K1 root rights, installed mainsail fluid timelaps and more via Guilouz from Github. Now I installed my camera today and want to take time-lapse photos. How does this work? What do I have to do because of everything I've tried so far It doesn't work. I use the Orca slicer as a slicer

Guilouz, 2024-01-31:
Timelapse have been tested and working fine.

Make sure to have command needed for Timelapse in your slicer by reading the documentation: https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Use-Timelapse

P-C-R, 2024-01-31:
> Timelapse have been tested and working fine.
> 
> Make sure to have command needed for Timelapse in your slicer by reading the documentation: https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Use-Timelapse

but why is then camera not showing in mainsail ?

[image]
[image]
[image]

Herbi88, 2024-01-31:
Have the same issue. Under Timelapse there is No camera to choose. Everything else is normal but when I look into folder no files are there. It worked with version 1.3.2.20 for me, but now it doesnt.