Guilouz/Creality-Helper-Script-Wiki #354: [BUG] guppy_module_loader

Badpioneer, 2024-02-03:
### What happened?

Hi!
During the "clean" installation of the new firmware v 1.3.3.5, and then an attempt to install using the Guppy screen script, I receive an error with the following content:
`{"code": "key341", "msg": "Section 'guppy_module_loader' is not a valid config section\n\nOnce the underlying issue is corrected, use the \"RESTART\"\ncommand to reload the config and restart the host software.\nPrinter is halted\n", "values": ["guppy_module_loader"]}`

### What did you expect to happen instead?

Correct Installation of Guppy Screen

### How to reproduce this issue?

1. Reset Factory Settings using this guide https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Reset-Factory-Settings
2. Install firmware v 1.3.3.5
3. Following first time calibration process
4. Get root access
5. Install helper script
6. Install Moonraker & Nginx                                        │
7. Install Fluidd
8. Install Klipper Gcode Shell Command  
9. Install Guppy Screen
10. Get error message

Badpioneer, 2024-02-03:
Additional info.
If you use the Guppy screen installation and repository command, then there are no problems. It looks like the reason is in the script
https://github.com/ballaswag/guppyscreen
`sh -c "$(wget --no-check-certificate -qO - https://raw.githubusercontent.com/ballaswag/guppyscreen/main/installer.sh)"`

Guilouz, 2024-02-03:
Just updated my script to support new version of Guppy Screen.

Please uninstall and reinstall it with latest script version.

closed 2024-02-03 by Guilouz