Guilouz/Creality-Helper-Script-Wiki #355: [BUG] <title>Script Helper install failed

Myklt8, 2024-02-03:
### What happened?

Get install failed error when sending command to run script

### What did you expect to happen instead?

Open Script Helper

### How to reproduce this issue?

install script helper 
run script helper

### Additional information:

[image]

Guilouz, 2024-02-03:
That's mean your printer can't downloading curl package with this command:

```
wget -q --no-check-certificate https://raw.githubusercontent.com/Guilouz/Creality-K1-and-K1-Max/main/Scripts/files/fixes/curl -O /tmp/curl
```

Myklt8, 2024-02-03:
I reset to factory again and it worked, but now can't install moonraker 
[image]

Myklt8, 2024-02-03:
I think it was because you were updating at the same time I was installing...all good now, thank you

closed 2024-02-03 by Myklt8

Guilouz, 2024-02-03:
I think you have issue with your connection.