Guilouz/Creality-Helper-Script-Wiki #358: [BUG] OBICO Install not working

P-C-R, 2024-02-03:
### What happened?

`Downloading Obico...Cloning into 'moonraker-obico'...
remote: Enumerating objects: 3233, done.
remote: Counting objects: 100% (1364/1364), done.
remote: Compressing objects: 100% (362/362), done.
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
./installer.sh: cd: line 2035: can't cd to moonraker-obico: No such file or directory
Installing dependencies and configuring Obico...sh: can't open './scripts/install_creality.sh': No such file or directory
`

### What did you expect to happen instead?

Obico install

### How to reproduce this issue?

install obico

### Additional information:

tried it 5 times with reboot. Internet connection seems fine

Guilouz, 2024-02-04:
Just installed Obico and working fine:

[image]

Issue is on your side.

closed 2024-02-04 by Guilouz

revunix, 2024-02-04:
A general question here about Obico, do you also only get 0.1 FPS with the camera?

P-C-R, 2024-02-04:
Yes. And I could no solve till now

@Guilouz I had to increase the http buffer from git from 1mb to 20mb. As we don't use ssh in the script the Obico install can fail then. Just FYI. Don't know if you want to add this before the install. (Obico is pretty big)

Guilouz, 2024-02-04:
> Yes. And I could no solve till now
> 
> @Guilouz I had to increase the http buffer from git from 1mb to 20mb. As we don't use ssh in the script the Obico install can fail then. Just FYI. Don't know if you want to add this before the install. (Obico is pretty big)

Strange I have installed it without issue

I can add git config --global http.postBuffer 20000000, that's you have applied ?

revunix, 2024-02-04:
> Yes. And I could no solve till now

That just doesn't make sense to me, because Octoeverywhere also works.

P-C-R, 2024-02-05:
> > Yes. And I could no solve till now
> > @Guilouz I had to increase the http buffer from git from 1mb to 20mb. As we don't use ssh in the script the Obico install can fail then. Just FYI. Don't know if you want to add this before the install. (Obico is pretty big)
> 
> Strange I have installed it without issue
> 
> I can add git config --global http.postBuffer 20971520, that's you have applied ?

are you using a ssh key on your test machine? nevertheless yes i would at least do it for the obico install