Guilouz/Creality-Helper-Script-Wiki #359: [BUG] Fluidd asking for password after reset

Galv87, 2024-02-04:
### What happened?

I had to reset my K1M to factory but after using the helper to reinstall fluidd it now asked for username and password but I have set none. How can I fix this? I can still access through mainsail port

### What did you expect to happen instead?

Straight log in to fluids no password required

### How to reproduce this issue?

Complete reset of the boot using a boot loader and then reinstall Fluidd on the same network

### Additional information:

It may be that the force login in set to true but I can’t access the moonraker file to change it

Guilouz, 2024-02-04:
Can't reproduce this issue. If you install moonraker first from script and then fluidd there is no issue, tested many times.

cjsieh, 2024-02-04:
Enable root and ssh in.

Add your subnet to  /usr/data/printer_data/config/moonraker.conf trusted_clients section. 

Or temporarily  enable your client host to be on one of these trusted subnets.   Then add your intended  subnet to moonraker.conf trusted clients.
    
The default moonraker has trusted users set to
trusted_clients:
  10.0.0.0/8
  127.0.0.0/8
  169.254.0.0/16
  172.16.0.0/12
  192.168.0.0/16
  FE80::/10
  ::1/128

This is standard moonraker trusted client code and not related to these helper apps.

closed 2024-02-04 by Guilouz