Guilouz/Creality-Helper-Script-Wiki #360: [BUG] fans with `[output_pin]` should ideally be using `[generic_fan]`

pedrolamas, 2024-02-04:
### What happened?

Most of the fans in the stock configurations are using `[output_pin]` should ideally be using `[generic_fan]`

### What did you expect to happen instead?

Fans configured with `[generic_fan]`

### How to reproduce this issue?

N/A

### Additional information:

This is actually not a bug, but I couldn't find a better way to report this.

For the record, I am opening this issue as a follow up on https://github.com/fluidd-core/fluidd/issues/1343

Guilouz, 2024-02-04:
Not an issue and not related to the wiki.

Creality has decided to use [output_pin] function rather than [fan_generic], it's a different approach but it works just as well.

Changing the name or function would break commands included in screen binaries whose sources are not available.

So we can't do anything about it.

closed 2024-02-04 by Guilouz

pedrolamas, 2024-02-04:
Fair enough, thank you for taking the time to reply!

Personally, this just gives me another reason to why we Fluidd devs (and same for Mainsail devs) lack the will to support anything related with Creality.